marcone/teslausb #878: DHCP isn't getting IP address

mulka, 2024-09-14:
### Describe the problem

I had a problem where the raspberry pi wasn't getting an IPV4 address for my home wifi so the archive loop was stuck in the `Waiting for archive to be reachable...` state. A bunch of other stuff seemed to be working though. For example, I was able to view footage on the Tesla screen. I was able to connect directly to the raspberry pi's wifi access point it was advertising and SSH into it.

When running `ifconfig` I got this for wlan0. Notice that there is no IPV4 address there.
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:feb4:cc9d  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:b4:cc:9d  txqueuelen 1000  (Ethernet)
        RX packets 562  bytes 51055 (49.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 160  bytes 31280 (30.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

I did some research with ChatGPT and found that I could manually run dhclient.
```
sudo -i
bin/remountfs_rw
dhclient -r wlan0
dhclient wlan0
```

Initial indications seem like that has worked. It is trying to archive a bunch of files now. Will wait to see if it completes and if the problem comes back in a few days.

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

mulka, 2024-09-16:
I did a little more investigation and found out that /etc/wpa_supplicant/wpa_supplicant.conf did not have the 
`id_str="AP1"` in it that is added `configure-ap.sh`. So, I don't know if something else maybe changed that file, or maybe there was a problem with the `configure-ap.sh` script and it didn't get to the end?

marcone, 2024-10-01:
> I did a little more investigation and found out that /etc/wpa_supplicant/wpa_supplicant.conf did not have the `id_str="AP1"` in it that is added `configure-ap.sh`. So, I don't know if something else [...]

Setup would have failed if configure-ap.sh failed. You can check /boot/teslausb-headless-setup.log to see if there were any issues during setup.

marcone, 2025-02-10:
Nothing actionable here

closed 2025-02-10 by marcone