marcone/teslausb #881: Pi zero not rebooting multiptle times after manual installating

Sjorsa, 2024-09-29:
### Describe the problem

I installed Raspberry Pi OS Lite (32bit) on a Pi Zero W using the official Raspberry Imager. 

After booting into that and updating with 'sudo apt update' and 'sudo apt upgrade', I run the installation command `curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash` as root and it does a whole bunch of stuff.

According to the instructions, the Pi should reboot multiple times now, but it only reboots once and doesn't do anything after that. 

If I run the installation command again, it returns `STOP: Previous resize attempt failed. Delete /root/RESIZE_ATTEMPTED before retrying.`. Also, there is no `/boot/teslausb-headless-setup.log` file created, so I cannot debug what is going wrong.

What could I be doing wrong?

### Device

Raspberry Pi Zero W

### OS Image

other

### Car Model

Model 3

### USB connection

Center console

marcone, 2024-09-29:
Note that the instructions say that the Pi _may_ reboot multiple times, not that it _will_ reboot multiple times.
Regardless, it looks like it either got stuck or was still busy doing the initial file system resize. How long did you wait before running the installation command again?

marcone, 2024-09-29:
OK, I can reproduce the problem. Must be something new/different about the current Raspberry Pi 32 bit lite image, since I installed this same way a few weeks ago. For now I'd recommend using the prebuilt TeslaUSB image.

marcone, 2024-09-30:
So far I've found that just flashing the "Raspberry Pi OS Lite (32 bit)" and then running the `curl ... | bash` install command works.
However if first I `apt update` and `apt upgrade`, then the install command fails as described above.

commit fff79309 (2024-10-01): setup: work around initramfs-tools bug

A recent update (or at least recent to Raspberry Pi OS. The change appears
to have occurred between 0.142 (which works) and 0.142+deb12u1 (which does
not work) to the initramfs-tools package made it so that copying an
indirectly symlinked host file to the initramfs would create a symlink
instead of copying the file if a file with the same name was already
indirectly symlinked in the initramfs.
Work around this by using the canonical name when copying.

Also make it so debian-resizefs.sh only updates the currently used
initramfs, to speed up setup.

fixes [... cut]
closed 2024-10-01 by marcone