marcone/teslausb #887: Audio skipping issue model Y

chamsters, 2024-10-16:
### Describe the problem

Hi all,

I have a Pi zero 2W with a 256GB U3 rated microsd that I'm connecting to the car through the glovebox USB slot.

Connecting the pi to my PC and running a usb speed test gives 30MB/s + sequential. 

I have no issues with Dashcam / sentry mode - both record fine.

Problem is with audio. I have 320kb MP3s, nothing too crazy but a decent collection. The car navigates the folders and plays the music but every so often (every couple of mins) it skips, like it's buffering. Can last a second, 2 seconds and resume. 

Any idea what the issue could be??

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

nilsmau, 2025-03-29:
i am experiencing the same thing ... basically music becomes unusable...
**Device**
Raspberry Pi Zero W

deangoldhill, 2025-06-25:
Yes I have noticed this too. It was perfect in the beginning and has got worse over time. Dashcam video files always have no problems.
I zero'd the card and reinstalled TeslaUSB and became perfect again, but a few months later same problem.
I suspect it's to do with continuously writing to the card and the way the file system works.
I use a SanDisk 128GB Extreme microSDXC, perhaps a different card? any suggestions?

THX723, 2025-06-26:
It might be worth checking your log and see if SSD TRIM is happening and successful.

On a non-related note, are any of you noticing the tracks end (cut off) several seconds earlier than it should be?

satanforaday, 2026-05-09:
Same skipping issue with a Raspberry Pi 4 Model B Rev 1.5. Was going to recreate my SD card but if other people are having this issue, don't see a fresh install is going to fix anything.