marcone/teslausb #890: TeslaUSB no longer recognised by car

jackmcconnell, 2024-10-21:
### Describe the problem

I have a UK spec 2019 Model 3 Performance with the latest firmware/software updates from Tesla. 
I'm using TeslaUSB on a Raspberry Pi Zero W with 128GB SD card. 
At somepoint in the past month, the car no longer recognises the Pi as a USB device. The Pi does power up and I can connect to Wifi on it though.
I tried a regular USB stick in it's place and it worked with the onboard dashcam and was detected by the car without issue. 
I wiped the SD card on the Pi and installed [the latest (pre-release) version of TeslaUSB](https://github.com/marcone/teslausb/releases/tag/v4) but this exhibts the same behaviour. 
Assuming Tesla hasn't nuked the ability to use TeslaUSB on a Pi via a software update in some way, what else can I try to get the car to recognise the Pi as a usb drive again?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Center console

marcone, 2024-10-21:
If you use the same cable that you use in the car to connect the Pi to a PC or laptop, does the PC/laptop see the Pi as a USB drive? (once it finishes booting up)

jackmcconnell, 2024-10-22:
@marcone Just checked when connecting to my Mac - no it doens't appear as a drive. It is accessible via Wifi though.
It wouldn't explain why it was working but then wasn't but does this mean I've missed something from the configuration file during intial set-up when I wiped it recently?

marcone, 2024-10-22:
Can you attach diagnostics here? You can either get them by logging in to the Pi and running `sudo /root/bin/setup-teslausb diagnose` or by connecting to the TeslaUSB web interface with a browser and downloading them from the diagnostics tab.

jackmcconnell, 2024-10-23:
No problem: [diagnostics.txt](https://github.com/user-attachments/files/17487271/diagnostics.txt)

ARAA2019, 2024-10-23:
I also have this issue. I was just gonna reinstall but maybe ill see what marcone says

marcone, 2024-10-23:
> No problem: [diagnostics.txt](https://github.com/user-attachments/files/17487271/diagnostics.txt)

As far as TeslaUSB is concerned it enabled the emulated drive, so the host (PC/car) should see it, assuming the USB cable you're using isn't a charge-only cable, and you have it attached to the correct port on the Pi (the one furthest away from the corner).

jackmcconnell, 2024-10-23:
That’s positive. I’ve only ever used a single micro-USB port on the Pi to both power it and for it to show be recognised as a USB drive in the Tesla. For this I’ve used the one on the far right. Can I use the one next to it to power it too?

marcone, 2024-10-23:
> That’s positive. I’ve only ever used a single micro-USB port on the Pi to both power it and for it to show be recognised as a USB drive in the Tesla. For this I’ve used the one on the far right. Can I use the one next to it to power it too?

I'm fairly certain you've never used the USB port on the far right (closest to the corner) and have TeslaUSB actually work as a USB drive, because that port (marked PWR on the circuit board) is only for power and doesn't have any data lines. You need to use the port labeled USB, as indicated here:
[image]
(you only need to use the USB port, as it carries both power and data)

jackmcconnell, 2024-10-23:
Well, stone me. You’re absolutely right. I must have messed this up at some point when re-wiring recently and not noticed. Entirely user error on my part. Thanks for your help in troubleshooting. Might be worth putting this in the docs for people like me 😂

marcone, 2024-10-23:
> Might be worth putting this in the docs for people like me

It already is, both in the [installation doc](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md#troubleshooting) and the [wiki](https://github.com/marcone/teslausb/wiki/Troubleshooting).

closed 2024-10-23 by marcone

jackmcconnell, 2024-10-24:
> > Might be worth putting this in the docs for people like me
> 
> It already is, both in the [installation doc](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md#troubleshooting) and the [wiki](https://github.com/marcone/teslausb/wiki/Troubleshooting).
I’m an idiot. Thank you!