marcone/teslausb #894: Retrying (1) 'archive_is_reachable' ... Using rclone and backblaze b2

Lumia1r, 2024-11-05:
### Describe the problem

Hi there,

Brand new user - but have tried to read up everything I can!
Plugged in today - all good, Car recognises drive and records fine.

Issue appears when parking at work and connecting to the wifi.
Folder appears on my b2 storage along with a handful of files. However i'm now just seeing..

`Waiting for archive to be unreachable...
Tue  5 Nov 10:09:40 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:09:41 GMT 2024: Retrying (1) 'archive_is_reachable' ...
Tue  5 Nov 10:09:42 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:09:43 GMT 2024: Retrying (2) 'archive_is_reachable' ...
Tue  5 Nov 10:09:44 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:09:45 GMT 2024: Retrying (3) 'archive_is_reachable' ...
Tue  5 Nov 10:10:01 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:10:02 GMT 2024: Retrying (1) 'archive_is_reachable' ...
Tue  5 Nov 10:10:03 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:10:04 GMT 2024: Retrying (2) 'archive_is_reachable' ...
Tue  5 Nov 10:10:05 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:10:06 GMT 2024: Retrying (3) 'archive_is_reachable' ...
Tue  5 Nov 10:11:00 GMT 2024: Sleeping before retry ...
Tue  5 Nov 10:11:01 GMT 2024: Retrying (1) 'archive_is_reachable' ...
Tue  5 Nov 10:11:23 GMT 2024: Sleeping before retry ...`

And no further files appearing on b2.
Wifi connection is strong, and connection must have been there initially for some files to appear. 
Sorry if this is dumb!

Thanks

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

marcone, 2024-11-05:
It might help if you post the entire log instead of a selected snippet.

Lumia1r, 2024-11-05:
Sorry about that - I re-ran /root/bin/setup-teslausb and the issue seemed to go away.

closed 2024-11-05 by Lumia1r