marcone/teslausb #901: Marker file does not get created for RecentClips

deangoldhill, 2024-11-24:
### Describe the problem

I am using https://github.com/ehendrix23/tesla_dashcam to merge clips,
So I have enabled the ARCHIVE_UPLOADED marker for all 3 Recent, Saved and Sentry clips
export TRIGGER_FILE_SAVED=ARCHIVE_UPLOADED
export TRIGGER_FILE_SENTRY=ARCHIVE_UPLOADED
export TRIGGER_FILE_RECENT=ARCHIVE_UPLOADED

Once the archive is complete, the marker is created for both Saved and Sentry clips, but not for RecentClips.
I get the telegram notification once complete, indicating the archive is succesful.

I did an upgrade just yeaterday, and still have the issue today
/root/bin/setup-teslausb upgrade

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

### Logs

Sun 24 Nov 09:41:54 GMT 2024: cleaning cam mount
Sun 24 Nov 09:41:54 GMT 2024: Ensuring cam file is mounted...
Sun 24 Nov 09:41:54 GMT 2024: Disconnecting usb from host...
Sun 24 Nov 09:41:54 GMT 2024: Running fsck on /backingfiles/cam_disk.bin...
Sun 24 Nov 09:41:54 GMT 2024: | fsck from util-linux 2.33.1
Sun 24 Nov 09:41:55 GMT 2024: | fsck.fat 4.1 (2017-01-24)
Sun 24 Nov 09:41:55 GMT 2024: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun 24 Nov 09:41:55 GMT 2024: |  Automatically removing dirty bit.
Sun 24 Nov 09:41:55 GMT 2024: | Long filename fragment "2024-11-13_18" found outside a LFN sequence.
Sun 24 Nov 09:41:55 GMT 2024: |   (Maybe the start bit is missing on the last fragment)
Sun 24 Nov 09:41:55 GMT 2024: |   Not auto-correcting this.
Sun 24 Nov 09:41:55 GMT 2024: | Performing changes.
Sun 24 Nov 09:41:55 GMT 2024: | /dev/loop1p1: 186 files, 333074/1572031 clusters
Sun 24 Nov 09:41:55 GMT 2024: Finished fsck on /backingfiles/cam_disk.bin.
Sun 24 Nov 09:41:55 GMT 2024: Running fsck on /backingfiles/music_disk.bin...
Sun 24 Nov 09:41:55 GMT 2024: | fsck from util-linux 2.33.1
Sun 24 Nov 09:41:56 GMT 2024: | fsck.fat 4.1 (2017-01-24)
Sun 24 Nov 09:41:56 GMT 2024: | /dev/loop1p1: 2106 files, 1956173/2094976 clusters
Sun 24 Nov 09:41:56 GMT 2024: Finished fsck on /backingfiles/music_disk.bin.
Sun 24 Nov 09:41:56 GMT 2024: Disconnected usb from host.
Sun 24 Nov 09:41:57 GMT 2024: Mounting /mnt/cam...
Sun 24 Nov 09:41:57 GMT 2024: Mounted /mnt/cam.
Sun 24 Nov 09:41:57 GMT 2024: Ensured cam file is mounted.
Sun 24 Nov 09:41:57 GMT 2024: Trimming free space in /mnt/cam, which has 279 extents
Sun 24 Nov 09:41:57 GMT 2024: Trim complete, image now has 279 extents
Sun 24 Nov 09:41:57 GMT 2024: Unmounting /mnt/cam...
Sun 24 Nov 09:41:57 GMT 2024: Unmounted /mnt/cam.
Sun 24 Nov 09:41:57 GMT 2024: done cleaning cam mount
Sun 24 Nov 09:41:58 GMT 2024: Finished archiving.
Sun 24 Nov 09:42:03 GMT 2024: Music archive not configured or unreachable
Sun 24 Nov 09:42:03 GMT 2024: unmounting /mnt/archive
Sun 24 Nov 09:42:03 GMT 2024: Connecting usb to host...
Sun 24 Nov 09:42:03 GMT 2024: Connected usb to host.
Sun 24 Nov 09:42:08 GMT 2024: Waiting for archive to be unreachable...
Sun 24 Nov 09:42:08 GMT 2024: Sleeping before retry ...
Sun 24 Nov 09:42:09 GMT 2024: Retrying (1) 'archive_is_reachable' ...
Sun 24 Nov 09:42:09 GMT 2024: Sleeping before retry ...
Sun 24 Nov 09:42:10 GMT 2024: Retrying (2) 'archive_is_reachable' ...
[... 166 lines cut ...]
Sun 24 Nov 15:33:50 GMT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000053
Sun 24 Nov 15:33:53 GMT 2024: took snapshot
Sun 24 Nov 15:33:53 GMT 2024: comparing new snapshot with /backingfiles/snapshots/snap-000052/snap.bin
Sun 24 Nov 15:33:53 GMT 2024: new snapshot is identical to previous one, discarding
Sun 24 Nov 16:31:54 GMT 2024: waiting up to 90 seconds for idle interval
Sun 24 Nov 16:33:30 GMT 2024: couldn't determine idle interval
Sun 24 Nov 16:33:30 GMT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000053
Sun 24 Nov 16:33:32 GMT 2024: took snapshot
Sun 24 Nov 16:33:32 GMT 2024: comparing new snapshot with /backingfiles/snapshots/snap-000052/snap.bin
Sun 24 Nov 16:33:32 GMT 2024: new snapshot is identical to previous one, discarding
Sun 24 Nov 17:31:33 GMT 2024: waiting up to 90 seconds for idle interval
Sun 24 Nov 17:33:08 GMT 2024: couldn't determine idle interval
Sun 24 Nov 17:33:09 GMT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000053
Sun 24 Nov 17:33:11 GMT 2024: took snapshot
Sun 24 Nov 17:33:11 GMT 2024: comparing new snapshot with /backingfiles/snapshots/snap-000052/snap.bin
Sun 24 Nov 17:33:11 GMT 2024: new snapshot is identical to previous one, discarding
==============================================

### Additional information

I have not changed the marker config since initial setup.
I have changed other config though (telegram setup).
Its not a major issue because I just use a cronjob to on the archive server to add the marker before the tesla_dashcam processing kicks in.

Thanks for your effrorts on this brilliant project!

closed 2024-11-24 by marcone

marcone, 2024-11-24:
Thanks for the report. Funny how apparently nobody else noticed in all the years since this feature was added.

jjermany, 2024-11-24:
Nice work guys. How do I update to this? I have custom wake_start and wake_stop scripts. If I update, will it affect them?

jjermany, 2024-11-24:
> Yes, if you run setup-teslausb upgrade then any changes you made to any
> of the TeslaUSB scripts or other code will be overwritten. This particular
> change was just one line, so you could easily just apply it yourself if you
> need it.
> You might also consider contributing your awake_start and awake_stop
> changes back to the project, if they are generally useful.

I can attach them here later and you guys can mess with it. It needs to be
rewritten to include other use cases. But it works great for people who use
TeslaFi.

deangoldhill, 2024-11-26:
working perfectly now since upgrade, marker files in all 3 directories post archive.
Thanks again for your efforts!

commit 49df42f2 (2024-11-24): Fix copy/paste error

RecentClips trigger file was being created in the SentryClips folder.
Fixes #901