marcone/teslausb #904: file-storage process hangs device

aneisch, 2024-12-21:
### Describe the problem

When attempting to copy files to device the `file-storage` process hogs the CPU and eventually causes the usb device to unexpectedly disconnect from my Mac or the vehicle. 

```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                           
 1192 root      20   0       0      0      0 R  97.0   0.0   3:52.22 file-storage 
```

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Glove box

### Logs

Script simply hangs and does not output results

aneisch, 2024-12-21:
I suspect this is just a kernel bug. I rebuilt the device using https://github.com/marcone/teslausb/releases/tag/v3.2 and it works as expected.

marcone, 2024-12-21:
> I suspect this is just a kernel bug. I rebuilt the device using https://github.com/marcone/teslausb/releases/tag/v3.2 and it works as expected.

which one did you use before?

aneisch, 2024-12-21:
I tried both https://github.com/marcone/teslausb/releases/tag/v5 and https://github.com/marcone/teslausb/releases/tag/v5.1

marcone, 2024-12-25:
I was able to reproduce this. First time I copied about 3GB worth of files to the CAM drive and that went OK; low CPU usage of the file-storage process, no messages in the kernel log, copy completed without error.
Then I tried another batch of files, and after a little while this started misbehaving:
started off with the same message in the kernel log you posted above, followed by this repeated over and over again:
```
[  698.621634] dwc2 3f980000.usb: new device is high-speed
[  698.690631] dwc2 3f980000.usb: new device is high-speed
[  698.742345] dwc2 3f980000.usb: new address 10
[  703.794460] dwc2 3f980000.usb: new device is high-speed
[  703.863481] dwc2 3f980000.usb: new device is high-speed
[  703.914857] dwc2 3f980000.usb: new address 10
[  708.959307] dwc2 3f980000.usb: new device is high-speed
[  709.031303] dwc2 3f980000.usb: new device is high-speed
[  709.082597] dwc2 3f980000.usb: new address 10
[  714.137180] dwc2 3f980000.usb: new device is high-speed
[  714.208152] dwc2 3f980000.usb: new device is high-speed
[  714.259205] dwc2 3f980000.usb: new address 10
```
The file-storage process was also using most of the CPU cycles at this point.

I canceled the copy on the Windows machine, but that didn't seem to work as the copy progress window just stayed opened.
I then logged in to the Pi and ran /root/bin/disable_gadget.sh
This disconnected the drive from the PC (which canceled the copy and dismissed the copy progress window), but it ended up hanging the script on the Pi. I suspect it was hanging on the `echo > "$gadget_root/UDC` since any access to `UDC` blocked at that point.

I haven't had a chance yet to try if it also fails like this with an official Raspberry Pi OS build.

[kern.log](https://github.com/user-attachments/files/18247351/kern.log)

marcone, 2024-12-26:
Same behavior with the official Raspberry Pi OS Lite 32-bit image.

marcone, 2024-12-26:
So far I've been unable to reproduce the behavior with the official Raspberry Pi OS Lite 64-bit image

marcone, 2024-12-26:
It looks to me like the problem occurs when the disk image gets near 4 GB in actual allocated size.
I can copy a little over 3.9 GB to the drive, but after that any attempts to copy additional files to the drive will fail in the manner described above, even after a reboot of the Pi.

I'm guessing the fact that it happens close to 4 GB is related to the fact that it only happens with the 32-bit image, though it's interesting that the problem doesn't appear to happen exactly at the 4 GB boundary, but slightly before.

marcone, 2024-12-26:
Unfortunately, kernel version 6.12, which is supposed to be the next stable kernel for Raspberry Pi OS, has the same problem.
I have a sneaking suspicion that [this code](https://github.com/raspberrypi/linux/blob/614fa9b0b1a21c0cc320b9915393bdaa31357de9/fs/iomap/buffered-io.c#L1141) is wrong, since there was a [previous change](https://github.com/raspberrypi/linux/commit/eee2d2e6ea5550118170dbd5bb1316ceb38455fb) that changed that same code construct because of a possible 32-bit overflow.

marcone, 2024-12-28:
It appears that the code I suspected is wrong is not actually the cause of the problem, since fixing it doesn't make the problem go away.
I have a fairly easy repro case using a standard Raspberry Pi OS image and not involving TeslaUSB or mass storage emulation, but I think the problem is an upstream linux kernel problem, not Raspberry Pi OS specific. Not sure where/how to report it though.

aneisch, 2024-12-28:
I appreciate and admire your deligance in tracking this down. I'm not sure where the best place to report would be either.

marcone, 2024-12-30:
It looks like there's already a discussion on the about the same/similar problem on the relevant kernel mailing list, so hopefully there will be a fix, and hopefully that fix will be backported to older affected kernels as well.

marcone, 2025-01-02:
I've sent a patch for the issue to the relevant Linux kernel mailing lists. We'll see if they take it, or if I get yelled at :)

marcone, 2025-01-07:
The patch is now in the upstream "linux-next" branch, from where it should eventually make its way into an upstream release.
It has also been merged into the 6.6, 6.12 and 6.13 branches for the official Raspberry Pi kernel, but it might take a while to see an actual new release from there.

Note again that this is only a problem for 32-bit kernels, so for devices with a 64-bit capable CPU you can use a 64-bit OS build. For Raspberry Pi Zero W (which is 32-bit only), please continue to use the TeslaUSB prebuilt image for now.

marcone, 2025-01-09:
Turns out that questionable code I mentioned in an [earlier comment](https://github.com/marcone/teslausb/issues/904#issuecomment-2563165101) is in fact also broken and can lead to a different 32-bit overflow and infinite loop. I've submitted a patch for that to upstream as well.

nowherenearaninja, 2025-01-11:
Yep, had this issue too.
First time user of teslausb, this is one of the coolest opensource projects I have come across. Thanks for your commitment.
The install process is awesome.

I had what i thought was usecase 1A, i.e. raspberry zero 2w, sd card only. Worked on first try, copied files to my NAS, but only bits. Thought it might be the speed of USB. But then eventually (a lot of headache) found this thread/issue. Reinstalled using raspian 64bit and then the full script instead of of prebuilt teslausb image. Et voila.

Suggestion if I may.
Add a temporary heads-up in the OneStepSetup doc that warns about using the prebuilt image, and instead point toward the full install process.

marcone, 2025-01-11:
> Suggestion if I may.
> Add a temporary heads-up in the OneStepSetup doc that warns about using the prebuilt image

I removed the prebuilt images that had the bug 2 weeks ago, so there is only one prebuilt image right now, which works.

closed 2025-02-04 by marcone