marcone/teslausb #907: Tesla’s new feature “Dashcam Clips” won’t work when using TeslaUSB

f4llenz, 2024-12-23:
### Describe the problem

[image]

Car is parked and charging, but not sleeping. The new feature makes sentry videos available for watching through the Tesla app. TeslaUSB still works fine as before, only having the “connect a faster USB” warning show up on the Tesla.

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

f4llenz, 2024-12-23:
Come to think of it, now that Tesla has a backend to expose the USB to the app, that might opens up another way to do what TeslaUSB does, but pulling it from the NAS/server rather than pushing from the raspberry pi. 

If I get time after the holidays I'll try to log the app network traffic

marcone, 2024-12-23:
You said "Car is parked and charging, but not sleeping", but not sleeping doesn't mean the USB ports are powered. It looks like maybe when you opened the "Dashcam Clips" view in the app, the Pi was still off or in the process of booting up. There isn't really anything TeslaUSB can do about this. You'll need to ensure the Pi is booted and connected to the car before using the new app feature.

closed 2024-12-23 by marcone