marcone/teslausb #908: Stuck Creating Partitions

TheAlphaLaw, 2024-12-30:
### Describe the problem

I had this working on my 2024 MS Plaid for about a week. I pulled the SD card out to adjust the cam size partition by adding an updated "teslausb_setup_variables.config" file. Since then, I'm unable to complete the install (with prebuild image).

**Toubleshooting:**
1. 10+ Fresh Installs\
2. Config File Adjustments
3. Different SD Cards (all over 256GB and new)
4. Mix of Pi Zero W 2's and Pi 4B's
5. Different PSU's
6. Different SSID's
7. Rebuilt Shared drive/folder (it's accessible on all other devices)
8. Manual Installs (Recieve SD Card to small error)

I'm at a loss at this point.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Glove box

### Logs

pi@teslausbModel-S:/boot $ tail -f teslausb-headless-setup.log
Sun 29 Dec 23:57:02 CST 2024 : Downloaded /root/bin/force_sync.sh ...
Sun 29 Dec 23:57:02 CST 2024 : Downloaded /root/bin/mountoptsforimage ...
Sun 29 Dec 23:57:02 CST 2024 : Downloaded /root/bin/mountimage ...
Sun 29 Dec 23:57:02 CST 2024 : Downloaded /root/bin/enable_gadget.sh ...
Sun 29 Dec 23:57:02 CST 2024 : Downloaded /root/bin/disable_gadget.sh ...
Sun 29 Dec 23:57:02 CST 2024 : cmdline.txt is up to date
Sun 29 Dec 23:57:08 CST 2024 : Downloaded /tmp/create-backingfiles-partition.sh ...
Sun 29 Dec 23:57:08 CST 2024 : create-backingfiles-partition: Checking existing partitions...
Sun 29 Dec 23:57:08 CST 2024 : create-backingfiles-partition: Modifying partition table for backing files partition...
Sun 29 Dec 23:57:08 CST 2024 : create-backingfiles-partition: Modifying partition table for mutable (writable) partition for script usage...

marcone, 2024-12-30:
>  I pulled the SD card out to adjust the cam size partition by adding an updated "teslausb_setup_variables.config" file

Just FYI, you can edit the config file on the Pi itself.

> lights blink ~6 times per second

if that's a series of blinks followed by a brief pause, then another series of blinks, etc. then that indicates the installer encountered an error. Not sure why the error doesn't show in the log file, but you can try ssh'ing into the Pi and running /root/bin/setup-teslausb manually to see if that shows more details.

TheAlphaLaw, 2024-12-30:
Thank you for the quick reply, here is what I see after manuall installing and reciving the "MicroSD card too small. . . " message:

```
root@teslausbModel-S:~#  /root/bin/setup-teslausb
Configuring the hostname...
Configured hostname: teslausbModel-S
/root/bin/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Set system timezone to America/Chicago.
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
Checking XFS support
XFS supported
DATA_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small: 3145728 bytes available.
Model: SD SN256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  266MB   262MB   primary  fat32        boot, lba
 2      266MB   2414MB  2147MB  primary  ext4
 3      2414MB  256GB   253GB   primary
```
Both lights are solid on the Pi4B.

TheAlphaLaw, 2024-12-30:
I've tried multiple power supplies and 3 brand new SD cards. 

There is nothing partitioned here to delete. Do I need to partition the drive?

```
Device         Boot   Start       End   Sectors  Size Id Type
/dev/mmcblk0p1 *       8192    520191    512000  250M  c W95 FAT32 (LBA)
/dev/mmcblk0p2       520192   4714496   4194305    2G 83 Linux
/dev/mmcblk0p3      4714497 499744767 495030271  236G 83 Linux
root@teslausbModel-S:~# sudo fdisk /dev/mmcblk0p3

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x1d33f8f3.

Command (m for help): p
Disk /dev/mmcblk0p3: 236 GiB, 253455498752 bytes, 495030271 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x1d33f8f3

Command (m for help): d
No partition is defined yet!
```

marcone, 2024-12-30:
You need to use `fdisk /dev/mmcblk0`, without the "p3" at the end

TheAlphaLaw, 2024-12-30:
Deleted, rebooted and same issue after manual run. 

```
`pi@teslausbModel-S:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 238.3G  0 disk
├─mmcblk0p1 179:1    0   250M  0 part /boot
├─mmcblk0p2 179:2    0     2G  0 part /
└─mmcblk0p3 179:3    0   236G  0 part
pi@teslausbModel-S:~ $  /root/bin/setup-teslausb
-bash: /root/bin/setup-teslausb: Permission denied
pi@teslausbModel-S:~ $ sudo -i
root@teslausbModel-S:~#  /root/bin/setup-teslausb
Configuring the hostname...
Configured hostname: teslausbModel-S
/root/bin/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Set system timezone to America/Chicago.
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
Checking XFS support
XFS supported
DATA_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small: 3145728 bytes available.
Model: SD SN256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  266MB   262MB   primary  fat32        boot, lba
 2      266MB   2414MB  2147MB  primary  ext4
 3      2414MB  256GB   253GB   `primary`
```

marcone, 2024-12-30:
Don't reboot after deleting the partition

TheAlphaLaw, 2024-12-30:
> Don't reboot after deleting the partition
> […](#)

If I do that then it tries to create a 4 partition and of course no sectors are available.

wartrichie, 2024-12-30:
Hey, I appear to be getting the same issue with a new Sandisk Ultra 256GB MicroSD, new cable and Raspberry Pi Zero following the one-step setup. It gets stuck creating partitions with the following error. I have tried deleting the 3rd partition and reformatting the MicroSD. Any help appreciated.

```
pi@teslausb:/ $ sudo /root/bin/setup-teslausb
/root/bin/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
Checking XFS support
XFS supported
DATA_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small: 3145728 bytes available.
Model: SD SD256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  266MB   262MB   primary  fat32        boot, lba
 2      266MB   2414MB  2147MB  primary  ext4
 3      2414MB  256GB   253GB   primary
```

TheAlphaLaw, 2024-12-30:
> Hey, I appear to be getting the same issue with a new Sandisk Ultra 256GB MicroSD, new cable and Raspberry Pi Zero following the one-step setup. It gets stuck creating partitions with the following [...]

I figured it was a bigger issue. I tried this with Sandisk SD Cards and Samsung. Same result.

marcone, 2024-12-30:
This should be fixed now.

closed 2024-12-30 by marcone