marcone/teslausb #909: Not recording continuously

Z33-6, 2025-01-06:
### Describe the problem

I have installed the 20211106.2 without any problems and it works absolutely fine for sentry recordings but doesn’t quite work properly for dashcam recordings. Correct me if I am wrong but I thought TeslaUSB was supposed to continuously record when driving so that I do not need to manually save any footage. On all of my journeys so far since the install, I might get one or 2 clips saved which will be around a few minutes. This is usually well into my journey, so nothing from beginning to end continuously, only usually a few clips towards the end of the journey. What could be wrong? When parked with the car own, it seems to record more often but during the actual drive, not so much

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

marcone, 2025-01-06:
Perhaps [SNAPSHOT_INTERVAL](https://github.com/marcone/teslausb/blob/51a95f9d9fe64740517a5384240c7bd486ae3d91/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L54) needs to be set for your location?

Z33-6, 2025-01-06:
How do I know what the interval needs to be for my location? I am in the UK that helps.

marcone, 2025-01-06:
You can start by just using the `export SNAPSHOT_INTERVAL=480` line from the current config file. If that doesn't solve the issue, please download diagnostics from the web interface and attach them here.

Z33-6, 2025-01-08:
So I used the default `export SNAPSHOT_INTERVAL=480` line and I think it may have made things worse. Yesterday it was just randomly recording 30-60 second clips a few times every hour. After a few hours of this, it would do nothing for the next few hours before going back to a few 30-60 seconds every hour. This was whilst the car was parked. I drove the car in the evening, doing two 30 minute trips and none of these trips were recorded. It looks like  the trips occured in the few hours of nothing gap mentioned above.

marcone, 2025-01-08:
Does the car show a red dot on the camera icon? It kind of sounds like the car isn't seeing the Pi's drive, or only intermittently.
If you can, leave the car parked within wifi range and with sentry mode enabled for a few hours, then download diagnostics from the web interface.

marcone, 2025-01-10:
I missed that you said it works fine for sentry but not for dashcam. Could it be that you have a loose connection somewhere or a bad cable, and vibration from driving is making the car repeatedly lose and regain the connection to the Pi?

Z33-6, 2025-01-10:
I did a few trips yesterday evening and again both of those trips did not get recorded. On one occasion I even pressed the dashcam icon on screen to manually saved the recording. This displayed a green check mark to confirm it had been saved but the recording is nowhere to be seen in TeslaUSB nor can I see it from the Tesla App under the new dashcam clips section. I have attached the logs so you can take a look.

marcone, 2025-01-11:
I don't see any obvious problems in the logs, though it's weird that when the free space manager deletes older snapshots to free up space that it sometimes deletes so many in a row. On my own device I see it delete up to 10 at a time to free up space, but in your logs I see it deleting up to 85 in a row.
When it's taking a new snapshot though, it always finds new files, so it does look like the car is continuously recording.

Where are you actually looking for these recordings and see them missing?

Z33-6, 2025-01-12:
I am looking for them under recent clips using the web ui. Yesterday I drove between 19:14 and 19:28 but its not showing in the recordings. Please see the screenshot attached to see that there is no recording that covers the time I was driving.

[image]

marcone, 2025-01-12:
Can you grab another diagnostics or archiveloop log and attach it here?

Z33-6, 2025-01-12:
Here is the archiveloop log file.

[archiveloop.log](https://github.com/user-attachments/files/18387878/archiveloop.log)

marcone, 2025-01-12:
So assuming that when you said that you drove yesterday between 19:14 and 19:28 you were talking about Friday the 10th of January, and that you're in the GMT time zone, here are the logs from around that period: 

```
Fri 10 Jan 19:16:30 GMT 2025: taking snapshot of cam disk in /backingfiles/snapshots/snap-000734
Fri 10 Jan 19:16:39 GMT 2025: took snapshot
Fri 10 Jan 19:16:40 GMT 2025: comparing new snapshot with /backingfiles/snapshots/snap-000733/snap.bin
Fri 10 Jan 19:16:40 GMT 2025: making links for /tmp/snapshots/snap-000734, retargeted to /backingfiles/snapshots/snap-000734/mnt
Fri 10 Jan 19:17:12 GMT 2025: made all links for /tmp/snapshots/snap-000734
Fri 10 Jan 19:17:36 GMT 2025: low space, deleting /backingfiles/snapshots/snap-000650
Fri 10 Jan 19:25:12 GMT 2025: waiting up to 90 seconds for idle interval
Fri 10 Jan 19:25:14 GMT 2025: write in progress
Fri 10 Jan 19:25:22 GMT 2025: no longer writing, wrote 132227072
Fri 10 Jan 19:25:28 GMT 2025: no writes seen in the last 5 seconds
Fri 10 Jan 19:25:28 GMT 2025: taking snapshot of cam disk in /backingfiles/snapshots/snap-000735
Fri 10 Jan 19:25:37 GMT 2025: took snapshot
Fri 10 Jan 19:25:39 GMT 2025: comparing new snapshot with /backingfiles/snapshots/snap-000734/snap.bin
Fri 10 Jan 19:25:39 GMT 2025: making links for /tmp/snapshots/snap-000735, retargeted to /backingfiles/snapshots/snap-000735/mnt
Fri 10 Jan 19:26:12 GMT 2025: made all links for /tmp/snapshots/snap-000735
Fri 10 Jan 19:26:24 GMT 2025: low space, deleting /backingfiles/snapshots/snap-000651
Fri 10 Jan 19:34:12 GMT 2025: waiting up to 90 seconds for idle interval
Fri 10 Jan 19:34:18 GMT 2025: write in progress
Fri 10 Jan 19:34:26 GMT 2025: no longer writing, wrote 132730880
Fri 10 Jan 19:34:32 GMT 2025: no writes seen in the last 5 seconds
Fri 10 Jan 19:34:32 GMT 2025: taking snapshot of cam disk in /backingfiles/snapshots/snap-000736
Fri 10 Jan 19:34:41 GMT 2025: took snapshot
Fri 10 Jan 19:34:43 GMT 2025: comparing new snapshot with /backingfiles/snapshots/snap-000735/snap.bin
Fri 10 Jan 19:34:43 GMT 2025: making links for /tmp/snapshots/snap-000736, retargeted to /backingfiles/snapshots/snap-000736/mnt
Fri 10 Jan 19:35:08 GMT 2025: made all links for /tmp/snapshots/snap-000736
```
so it took 3 snapshots that cover some part of your drive, and each snapshot differed from the previous one, so it looks like the car was recording during that time (though technically it's possible the car didn't record anything new and only deleted previously recorded stuff).
These snapshots were deleted as part of a mass delete of 78 snapshots on Saturday starting at 12:27, which is why you will no longer see those recordings now.
Can you ssh into the Pi and do:
```
sudo -i
mount /mnt/cam
df -h /mnt/cam
umount /mnt/cam
```
and reply here what it says?

Z33-6, 2025-01-12:
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/loop0      120G   46G   75G  38% /mnt/cam
```

marcone, 2025-01-13:
That looks fine. I am frankly at a loss as to why your setup is behaving the way it is. I will try adding some more logging to debug further.

Z33-6, 2025-01-13:
I’ll just reinstall it and see if that sorts this issue out

marcone, 2025-01-16:
> I’ll just reinstall it and see if that sorts this issue out.
> 
> Just to be clear, the dashcam is expected to continuously record without the need to manually (using horn or pressing icon) save the clips right?

Dashcam should record whenever the car is on. See [here](https://github.com/marcone/teslausb/wiki/How-a-Tesla-records-Dashcam-&-Sentry-footage) for more details.

> Also, should the dashcam clips not stay on the sd card for longer than a few days? Its a 256gb card of which 120gb was selected for cam

TeslaUSB will keep older recordings until the space is needed for newer recordings. I would suggest setting CAM_SIZE a lot smaller than 120 GB though, unless you have a specific reason for it to be that large.

Did reinstalling fix the issue?

marcone, 2025-01-17:
> I thought the more space I give it then the more recordings it can hold at once

That's now how TeslaUSB works. The CAM_SIZE you specify is the size of the drive that the car will see. Within that drive, the car will still only keep one hour of dashcam footage. To keep more than an hour, TeslaUSB needs to copy the older footage before the car deletes it. TeslaUSB needs space to store those copies, but if all the space is already used (because you specified 120G or even 200G for the drive size) then TeslaUSB can't do that.

Z33-6, 2025-03-02:
Apologies for the late reply - life got in the way so this fell to the bottom of my list. Having reinstalled it from scratch, the same issues were still occuring. One thing I did notice under my notifications in the Tesla screen was a message that said: “Some clips not saved - USB too slow for Dashcam. Use USB drive that writes at 4 MB/s or higher“. Do you think this could be the reason why some clips are missing? The SD card in the pi is a SanDisk endurance card made for dashcams so that is not the problem. I have tried different cables as well to eliminate a faulty or slow cable however the message is still appearing. Is there something I could do or is this because of a limitation with the Pi Zero 2 W? If it is the latter, then I’m not sure why it is considered a supported device. Your advice would be greatly appreciated.

marcone, 2025-03-02:
> The SD card in the pi is a SanDisk endurance card made for dashcams so that is not the problem

I don't think that's necessarily a foregone conclusion. High endurance cards tend to be slower than regular SD cards, though since the car does not record directly to the Pi's SD card, that may not matter as much. Still, if the card is extremely slow or defective, it may stall the Pi's writes.
The Raspberry Pi Zero 2W's USB drive emulation should run at up to about 20 MB/sec write speed. You can measure what speed you're getting by plugging the Pi into a PC and running a speed test on the "CAM" drive.

Z33-6, 2025-03-03:
I ran the dd command on CAM to test the read and write speed several times, and for both operations the speed varied between 3MB/s and 74MB/s

marcone, 2025-03-03:
Sounds like you're not measuring the correct thing(s), as the RPi Zero 2W OTG emulation is not going to be significantly faster than 20 MB/s, so if you're seeing 74 MB/s, you're probably measuring speed to the host cache or something.
I prefer to connect the Pi to a Windows machine and using CrystalDiskMark to measure the disk speed. (if using Windows, note [this](https://github.com/marcone/teslausb/wiki/FAQ#q-i-plugged-the-pi-into-a-windows-pc-and-copied-a-large-file-to-it-and-it-told-me-its-only-writing-2mbsec) though)

Z33-6, 2025-03-03:
I used CrystalDiskMark and these are my results: [https://ibb.co/Mkfc7GKx](url)

That doesn’t match up to the slow i/o speeds that the Tesla is warning me about.

Z33-6, 2025-03-06:
Any thoughts on why I am getting slow i/o speeds and if that is the reason why some of my recordings are missing?

marcone, 2025-03-06:
> Any thoughts on why I am getting slow i/o speeds and if that is the reason why some of my recordings are missing?

I don't know. I've never even seen the message you mentioned. I know the car will display a warning when plugging a USB2 device (like the Pi) into the glovebox USB port, however that doesn't prevent it from working. I've never seen a message explicitly saying that some clips were not recorded _because_ the usb drive is too slow, as the message you quoted seems to be saying.

Z33-6, 2025-03-07:
[image]

This is the exact message that gets shown

jjjasont, 2025-03-25:
@Z33-6 ,
I also have the same error message as you do. and every now and then the clips are not saved in teslausb.

I added `dwc_otg.speed=0` to the end of the cmdline.txt file and so far (today ~4 hours of driving) there is no missing recording. Give this a try, see if it helps.

Z33-6, 2025-03-31:
@jjjsont, thanks will give this a code - but should it not be dwc_otg.speed=1, not dwc_otg.speed=0?

jjjasont, 2025-03-31:
Per https://www.cl.cam.ac.uk/~atm26/ephemeral/rpi/dwc_otg/doc/html/module%20parameters.html
speed = 0 is high speed (480 Mbps)
speed =1 is full speed (12 Mbps)

Although by default pi would use high speed, but every now and then Tesla will mount my TeslaUSB Full-speed. Causing the clips not saved in full. So I'm forcing the high speed option here.

Combined with SNAPSHOT_INTERVAL=480, Tesla has not missed any recordings so far, when pi is correctly mounted.

There has been 1 times where Tesla unable to recognize TeslaUSB half way the drive. (At the beginning of the drive, I see a red circle. Half way the trip, it changed to a red cross. It fixes itself after a reboot.) But at least when I see red circle, TeslaUSB is saving the recordings.

Z33-6, 2025-04-04:
Just gave it a try yesterday and drove for around 6 hours. Unfortunately I was still getting the same error message on screen. This wasnt a random message, it was happening quite often.