marcone/teslausb #921: 43455 wifi chipset can't do 2.4 and 5 at the same time, but NetworkManager seems to try anyway, breaking STA side

zxcv-org, 2025-01-25:
### Describe the problem

On a pi5, it seems like NetworkManager can select 2.4GHz for AP side while STA side is using 5GHz. However, the 43455 wifi chipset can't do this, as it requires same band and same channel for the AP and STA (see link below confirming this).

Initial repro:

I enabled the AP in the teslausb config, and told the STA side to connect to a 5GHz-only AP. The STA would sometimes almost completely fail (just a few packets every ~15s or so seems like). Using nmcli d wifi list (IIRC), I noticed that the AP side had selected 2.4GHz instead of 5GHz, while the STA side was still using 5GHz. For completeness, initially STA was using 5GHz because of 5GHz-only SSID, and later STA was using 5GHz because of band=a in the STA nmconnection file under [wifi], but with no "band" in the AP nmconnection file.

Maybe faster repro (speculative):

It may be faster to repro the symptom by forcing the STA to 5GHz using nmconnection [wifi] band=a (5GHz), and force the AP to band=bg (2.4GHz) in its file. However, I haven't tried this so YMMV.

### Device

Raspberry Pi 5

### OS Image

other

### Car Model

Model 3

### USB connection

Center console

### Logs

no relevant teslausb logs that I know of since this is purely a wifi connectivity issue

### Additional information

I started with the raspberry pi imager raspbian 64 bit lite option, upgraded packages, and installed teslausb using the full install steps (not the pre-built image), and un-commented the AP config lines etc.

The following link clarifies that 43455 wifi chipset is not capable of doing STA and AP on two different bands or two different channels at the same time. The band and channel must be the same:

https://community.infineon.com/t5/Wi-Fi-Combo/Looking-for-details-regarding-CYW43455-concurrent-Station-and-AP-operation/td-p/122651#.

AFAICT, the created NetworkManager nmconnection files don't nail down which band (which would normally be fine on chips that can do simultaneous dual band), and it seems that NetworkManager can decide (sometimes) to use 2.4GHz for the AP even though the STA side is using 5GHz.

As far as I can tell, at that point, the STA side can go into a mode where maybe a couple pings every 15 or so are responded to, the first with latency over a second, the second quickly, but then it goes back to not responding to pings for a while. Connections on the STA side are pretty much impossible when in this failure mode. I'm not sure if this could potentially happen on the AP side randomly without me noticing, as I don't use the AP side as much. I did test during one of the STA outages that the AP side worked fine at the same time as the STA outage. So it seems like the AP side can "win" sometimes I guess.

IIUC, the nmconnection file [wifi] band=a parameter locks to 5GHz, so I just locked both the STA side and AP side to 5GHz, which seems to work for my setup (so far at least - hard to know how long to wait before filing this). I didn't experiment with locking both to 2.4GHz but maybe that would also work fine.

I'm not sure what's the best way to mitigate this issue. It seems like NetworkManager doesn't know that with the 43455, NetworkManager would/could/should need to ensure the band and channel match for the STA and AP on the same 43455 chip. Or maybe something below NetworkManager, but anyway.

Maybe just a note in the teslausb setup guides for now re. potentially adding "band=a" or "band=bg" to both STA and AP, at least until more folks have confirmed the same failure mode and fix.

Based only on my ad-hoc troubleshooting and best-effort observations and limited testing so far, it seems possibly premature to offer a config setting to lock both to 5 or both to 2.4, but your call.

I'm not sure if this only impacts the pi5, but that's what I'm using. IIUC the pi4 also uses the 43455; no idea if it can hit the same problem.

I had previously tried what seemed like later firmware for the 43455, and while that firmware did "just work" (surprising me), the failure mode still happened with that FW. I didn't try any brcmf... driver changes.

The two observations that convinced me to open this "issue":
  * Once, when I re-enabled the AP side explicitly (I don't recall which nm command), the failure mode _immediately_ kicked in.
  * The observation that AP side had picked 2.4GHz as seen in "nmcli d wifi list" output, despite the STA side only connecting to a 5GHz-only SSID.

Unfortunately I don't have a record of eeprom firmware updates I've done or any wifi-related logs, but if there are any commands I can run to gather more info to help narrow down the scope of the failure mode (in case it only happens under some setups), please let me know.

zxcv-org, 2025-01-26:
Well, it happened again eventually despite setting both to 5GHz... (with very few ping responses showing up on a periodic basis)

And now after moving the AP nmconnection file out of view of NetworkManager, another outage with no ping responses at all. Maybe this is a different problem since the symptom is different.

I'll comment again if I can narrow it down further/better.

I've seen zero wifi STA issues on a pi5 running ubuntu... So ... at least there's hope that some sort of SW change somewhere could maybe help.

zxcv-org, 2025-01-26:
I've put an AP closer to the pi5 to improve the signal strength quite a bit. I think that was likely the main problem, despite the sus time correlation around activating the AP side.

closed 2025-01-26 by zxcv-org

marcone, 2025-01-26:
For what it's worth, I have one Raspberry Pi Zero 2 W that doesn't seem able to reliably do simultaneous AP and client mode. Works fine on my other RPiZero2W though. Same software, same AP, same location. One works, the other doesn't.