marcone/teslausb #923: OrangePi Zero 2W install steps

stewartie4, 2025-01-26:
### Describe the problem

On a reddit thread someone mentioned that an Orange Pi Zero2 W would be preferable to a Raspberry Pi Zero 2w as the former has USB-C support and more RAM.
I bought an OrangePi with that in mind, naively thinking it would be a drop in replacement for a RPI.
I've since found that this is obviously not the case (most RBI images including the prebuilt image don't boot) and spent days trying to get an image/script combo to work with teslausb.

I eventually got everything working so posting here for posterity - **obviously just buy a RPI if you don't want the hassle**

Steps:
Image: Orangepizero2w_1.0.2_debian_bookworm_server_linux6.1.31 (5.x kernal and ubuntu imsges don't boot properly or work post-teslacam setup)
Burnt with: Balena Etcher 

**1. Setup ssh:**
- Unfortunately these images don't have a first boot wifi config, so you'll need to connect the orangepi to ethernet, or connect it to a monitor and configure wifi via orangepi-config or nmcli.
- Once networking is set up, find your pi's ip address (via your router admin page or unix commands)
- Then run ssh orangepi@<ip> (password is orangepi)
From here you can begin setup.

**2. Install missing dependencies**
- Switch to the root user (sudo su -)
- Update all dependencies/repos as some repos are unhealthy and will cause issues in the teslausb setup script (apt update)
- Install dos2unix which is missing and required in the scripts (apt install dos2unix)
- Enable xdc module (sudo modprobe xfs)
- Enable xdc on boot (echo xfs | tee -a /etc/modules)

**3. Setup wifi**
- Setup wifi (see step 1) even if you are on wired as it will be more difficult to do this on the read only filesystem after install 

From here you can follow the existing steps for "other SBCs" [here](https://github.com/marcone/teslausb/wiki/Installation)

Thank you for this amazing project!

### Device

Other

### OS Image

other

### Car Model

Model 3

### USB connection

Glove box

marcone, 2025-01-26:
> On a reddit thread someone mentioned that an Orange Pi Zero 2W would be preferable to a Raspberry Pi Zero 2W for this project as the former has USB-C support and more RAM. I bought an OrangePi with that in mind, naively thinking it would be a drop in replacement for a RPI. I've since found that this is obviously not the case (most RBI images including the prebuilt image don't boot)

"most" images, or *all* images? I wouldn't expect any Raspberry Pi image to boot on Orange Pi, because it's different hardware.

> and I have spent days trying to get an image/script combo to work with teslausb.
> 
> I eventually got everything working so posting here for posterity - **obviously just buy a RPI if you don't want the hassle**
> 
> Steps: Image: Orangepizero2w_1.0.2_debian_bookworm_server_linux6.1.31 (5.x kernel and ubuntu images don't boot properly or work post-setup) Burnt with: Balena Etcher
> 
> **1. Setup ssh:**
> 
> * Unfortunately these images don't have a first boot wifi config, so you'll need to connect the orangepi to ethernet, or connect it to a monitor and configure wifi via orangepi-config or nmcli.
> * Once networking is set up, find your pi's ip address (via your router admin page or unix commands)
> * Then run ssh orangepi@{ip address} (password is orangepi)
>   From here you can begin setup.
> 
> **2. Install missing dependencies**
> 
> * Switch to the root user (sudo su -)
> * Update all dependencies/repos as some repos are unhealthy and will cause issues in the teslausb setup script (apt update)
> * Install dos2unix which is missing and required in the scripts (apt install dos2unix)

that shouldn't be necessary as the TeslaUSB manual setup script installs it.
> * Enable xdc module (modprobe xfs)
> * Enable xdc on boot (echo xfs | tee -a /etc/modules)

I assume "xdc" is just a typo for "xfs"? 

> **3. Setup wifi**
> 
> * Setup wifi (see step 1) even if you are on wired as it will be more difficult to do this on the read only filesystem after install
> 
> From here you can follow the existing steps for "other SBCs" [here](https://github.com/marcone/teslausb/wiki/Installation)
> 
> Thank you for this amazing project!
> 
> ### Device
> OrangePi Zero 2W
> 
> ### OS Image
> Orangepizero2w_1.0.2_debian_bookworm_server_linux6.1.31

I'm going to close this since it is not a bug/issue, but feel free to repost in the discussion forum, or add a page to the wiki similar to what I did for the Orange Pi Zero 3.

closed 2025-01-26 by marcone

stewartie4, 2025-01-26:
> that shouldn't be necessary as the TeslaUSB manual setup script installs it.
It doesn't seem to - i was getting STOP dos2unix command not found

>I assume "xdc" is just a typo for "xfs"?
Thanks, typo

> add a page to the wiki similar to what I did for the Orange Pi Zero 3.
Do'h! google did not show that [wiki](https://github.com/marcone/teslausb/wiki/Orange-Pi-Zero-3-Installation)! I assume because it references the Orange Pi 3 not 2W but the steps should work nonetheless! 
I wish I had seen it a few days ago, it would have saved me a lot of trial and error!

Still hopefully this issue will be picked up by the google spider and help someone in future

marcone, 2025-01-26:
> > that shouldn't be necessary as the TeslaUSB manual setup script installs it.
> 
> It doesn't seem to - i was getting STOP dos2unix command not found, it also doesn't handle loading the xfs kernal module and setting it up on boot

What command/script were you running? The install.sh script listed on the manual install page that you referenced above will [install dos2unix](https://github.com/marcone/teslausb/blob/df02817e5c4edf236c6e1a6ac10cd4b61cec34b2/setup/generic/install.sh#L176) as well as a few other dependencies.