marcone/teslausb #941: Error during archiving: "Archived 0 files in 2s"

Gizmotoy, 2025-03-10:
### Describe the problem

I'm pretty frequently encountering an issue where TeslaUSB begins archiving and then fails immediately.  I use PushBullet, and get the following message:
"Error curing archiving.  Archived 0 files in 2s"
I would say this happens every other time the vehicle attempts to archive.

Looking at my log, this is the lead-up to that message:
```
Sun  9 Mar 11:43:18 PDT 2025: Archiving...
Sun  9 Mar 11:43:18 PDT 2025: Ensuring cam archive is mounted...
Sun  9 Mar 11:43:18 PDT 2025: Mounting /mnt/archive...
Sun  9 Mar 11:43:18 PDT 2025: Mounted /mnt/archive.
Sun  9 Mar 11:43:18 PDT 2025: Ensured cam archive is mounted.
Sun  9 Mar 11:43:18 PDT 2025: Checking saved folder count...
Sun  9 Mar 11:43:20 PDT 2025: There are 1 event folder(s) with 47 file(s) and 0 track mode file(s) to move.  0 short recording(s) will be skipped.
Sun  9 Mar 11:43:20 PDT 2025: Starting recording archiving
Sun  9 Mar 11:43:20 PDT 2025: Archiving 47 file(s) including 1 event folder(s) starting at Sun  9 Mar 11:43:20 PDT 2025
Sun  9 Mar 11:43:20 PDT 2025: Sending Pushover message.
Sun  9 Mar 11:43:22 PDT 2025: Error during archiving. Archived 0 files in 2s
Sun  9 Mar 11:43:22 PDT 2025: Sending Pushover message.
2025/03/09 11:43:21 [3523] building file list
2025/03/09 11:43:21 [3523] done
building file list ... done
rsync: [Receiver] ERROR: cannot stat destination "/mnt/archive": Stale file handle (116)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(765) [Receiver=v3.2.3]
```

This began around the time the issue where only 2 files were being saved was resolved.  I have two separate vehicles running into this issue on two different Pi Zero W (both using the pre-built image).

Any idea what might be going on?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Center console

marcone, 2025-03-10:
Googling around a bit for that error message I found that adding `noserverino` to the mount options may fix the problem. Can you give that a try?
E.g. something like:
```
sudo -i
bin/remountfs_rw
sed -i "s/cifs rw/cifs rw,noserverino/" /etc/fstab
```

Gizmotoy, 2025-03-11:
@marcone 
Ok, made the change and so far so good.  No instances since the change and I'll keep an eye on it.

I also noticed this in the logs.  Is there a way to manually run fsck on `cam_disk.bin` to delete these fragments?  I see it on both of my instances.

```
Sun  9 Mar 19:27:19 PDT 2025: Running fsck on /backingfiles/cam_disk.bin...
Sun  9 Mar 19:27:19 PDT 2025: | fsck from util-linux 2.33.1
Sun  9 Mar 19:27:27 PDT 2025: | fsck.fat 4.1 (2017-01-24)
Sun  9 Mar 19:27:27 PDT 2025: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  9 Mar 19:27:27 PDT 2025: |  Automatically removing dirty bit.
Sun  9 Mar 19:27:27 PDT 2025: | Long filename fragment "2024-07-04_12" found outside a LFN sequence.
Sun  9 Mar 19:27:27 PDT 2025: |   (Maybe the start bit is missing on the last fragment)
Sun  9 Mar 19:27:27 PDT 2025: |   Not auto-correcting this.
Sun  9 Mar 19:27:27 PDT 2025: | Long filename fragment "2024-05-08_08" found outside a LFN sequence.
Sun  9 Mar 19:27:27 PDT 2025: |   (Maybe the start bit is missing on the last fragment)
Sun  9 Mar 19:27:27 PDT 2025: |   Not auto-correcting this.
Sun  9 Mar 19:27:27 PDT 2025: | Performing changes.
Sun  9 Mar 19:27:27 PDT 2025: | /dev/loop2p1: 3487 files, 3028832/3275967 clusters
```

Gizmotoy, 2025-03-14:
Checking back in: zero recurrences of this since on either vehicle making that change.

closed 2025-03-15 by Gizmotoy