marcone/teslausb #942: Archiving Error - Broken pip??

vical22, 2025-03-11:
### Describe the problem

It has been working perfectly, but I noticed the archiving stopped.

Had a look at the archiving log, and found there has been an error during archiving, and it gives "rsync: [sender] write error: Broken pipe (32)" and [rsync](rsync: [receiver] close failed) on (path).

Not sure what is happening.

Should I rebuild it from scratch?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

### Logs

[archiveloop.log](https://github.com/user-attachments/files/19180594/archiveloop.log)

[diagnostics.txt](https://github.com/user-attachments/files/19180596/diagnostics.txt)

marcone, 2025-03-17:
Are  you using an unraid server by any chance? If so, see https://github.com/marcone/teslausb/issues/736

polzovotel, 2026-08-29:
I was getting repeated `rsync: write error: Broken pipe (32)` during TeslaUSB archiving on a Raspberry Pi Zero 2 W running Debian 13. 

This started to happen after 2 weeks trip that accumulated lot of sentry events. The archive server showed `Connection reset by peer`, and `journalctl -k` on the Zero showed the OOM killer was killing `ssh`. Rsync itself was only using ~8 MB RAM. 

The real issue was that the Pi had 256 MB reserved for CMA out of only ~426 MB usable RAM. I added `dtoverlay=cma,cma-64` under `[all]` in `/boot/firmware/config.txt` and rebooted. CMA dropped to 64 MB and the rsync archive has been stable since.