marcone/teslausb #953: Setup stopped when create-backingfiles-partition being formatted

blacktiger333, 2025-04-28:
### Describe the problem

tried now with several micro-sd cards from different vendors and 2 different USB cables to setup teslausb for my Raspberry Pi Zero W with TeslaUSB image Pre-release from 20250203, but also tried with TeslaUSB image teslausb-20211106.2 (latest). Setup always stops when create-backingfiles-partition is being formatted (see log file). Any idea what's wrong?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Center console

### Logs

==> /teslausb/teslausb-headless-setup.log <==
Mon Apr 28 05:14:46 PM CEST 2025 : Updated cmdline.txt from:
Mon Apr 28 05:14:46 PM CEST 2025 : console=serial0,115200 console=tty1 root=PARTUUID=440f1415-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US
Mon Apr 28 05:14:46 PM CEST 2025 : to:
Mon Apr 28 05:14:46 PM CEST 2025 : console=serial0,115200 console=tty1 root=PARTUUID=440f1415-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US modules-load=dwc2,g_ether
Mon Apr 28 05:14:51 PM CEST 2025 : Downloaded /tmp/create-backingfiles-partition.sh ...
Mon Apr 28 05:14:51 PM CEST 2025 : create-backingfiles-partition: Checking existing partitions...
Mon Apr 28 05:14:51 PM CEST 2025 : create-backingfiles-partition: Modifying partition table for backing files partition...
Mon Apr 28 05:14:52 PM CEST 2025 : create-backingfiles-partition: Modifying partition table for mutable (writable) partition for script usage...
Mon Apr 28 05:14:52 PM CEST 2025 : create-backingfiles-partition: Writing updated partitions to fstab and cmdline.txt
Mon Apr 28 05:14:52 PM CEST 2025 : create-backingfiles-partition: Formatting new partitions...

====================================

### Additional information

[diagnostic.log](https://github.com/user-attachments/files/19942829/diagnostic.log)

marcone, 2025-04-28:
Your log shows an an error when writing to the SD card. Since you said you tried multiple SD cards, it seems they were either all bad (assuming they all failed in the same way), or you have a bad Pi.

marcone, 2025-05-15:
Can't really do anything about hardware problems.

closed 2025-05-15 by marcone