marcone/teslausb #958: BLE pairing doesn't work with prebuilt image

n0k0m3, 2025-05-12:
### Describe the problem

Prebuilt image doesn't have bluez firmwares and can't find bluetooth hci
```
sudo /root/bin/tesla-control -ble -vin $VIN add-key-request /root/.ble/key_public.pem owner cloud_key
Error: ble: failed to enable device: can't init hci: no devices available
```
```
hcitool dev
Devices: #Empty
```

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

[diagnostics.txt](https://github.com/user-attachments/files/20165001/diagnostics.txt)

### Additional information

The log above is AFTER installing bluez, bluez-firmware, libbluetooth3.

marcone, 2025-05-12:
Did you reboot after installing bluez and those other packages?

n0k0m3, 2025-05-12:
Yes I did.
For more information I'm running the 2023 latest release image not the 2025 latest pre-release. Testing the pre-release rn

n0k0m3, 2025-05-12:
pre-release doesn't boot, moved to pios bookworm lite image + install manually and the feature worked (bluetooth hci recognized)

jakest, 2025-06-26:
I couldn’t get it working with the prebuilt image. I installed Raspbian Lite then used the install script. Once that was done, I had to manually install `bluez`, `bluez-firmware`, and `libbluetooth3`.

But even so, the logs say that BLE can’t be used to enable to disable Sentry, even though it’s showing connected as a key. I haven’t been able to get it working at all.

jakest, 2025-07-03:
Pi Zero 2 W

Yes, I have a phone and watch connected. I actually think I have two watch apps connected (Tesla and Tessie). I’ve had more than that in the past, but those are the only two on the list now.

THX723, 2025-07-03:
> Meaning concurrent ever? Or concurrently active? I can definitely try removing the watch apps. If concurrent, my watch was out of range during this testing (I typically don't wear it on the weekend). VIN is correct, but I'll give it another shot by just removing one of the watch apps so there are only three concurrent connections.

Concurrently active. Powering down the watch is the surest way to be sure it is not connected.

THX723, 2025-07-03:
Btw Key Fob, if you have any, also counts as one.

THX723, 2025-07-03:
> > There's a 3 concurrent BLE connection limit. Try powering down your watch and see if that helps. Also double check your VIN entry, and also make sure they are in CAPS.
> 
> I am not sure if this is fully correct. I have 2 phones, 1 watch, 1 charger BLE connected to the car. 
> 

A BLE charger? That's a new one! :)

You are confusing BLE with standard BT. They are unique classes of Bluetooth. BLE doesn't typically maintain a persistent connections like standard BT. The two can coexist on top of each other.