marcone/teslausb #966: Fix: after boot or "Archive is reachable" firmware errors UI_a112_badUSBDevice

mhaluska, 2025-06-09:
### Describe the problem

This fully fixed my Tesla firmware errors "UI_a112_badUSBDevice" immediately after RPi boot or when I returned home = Archive reachable.

```
--- bin/archiveloop.bak 2025-06-06 21:06:40.813614987 +0200
+++ bin/archiveloop     2025-06-09 18:29:07.091652092 +0200
@@ -304,6 +304,7 @@
 
 function disconnect_usb_drives_from_host () {
   log "Disconnecting usb from host..."
+  /root/bin/waitforidle || true
   if /root/bin/disable_gadget.sh
   then
     fix_errors_in_images
@@ -907,6 +908,7 @@
   # take a snapshot before archive_clips starts deleting files
   if has_cam_disk
   then
+    /root/bin/waitforidle || true
     /root/bin/make_snapshot.sh
   fi
 
```

### Device

Raspberry Pi 5

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2025-06-13:
This makes no sense to me, but in any case, code changes should be submitted as a [pull request](https://github.com/marcone/teslausb/pulls), not as a diff in an issue.

closed 2025-06-13 by marcone

mhaluska, 2025-06-13:
First is for not disconnecting USB drive during write = prevent UI_a112_badUSBDevice. Second for consistency of snapshot, there can be write in progress. Does this make sense?

THX723, 2025-06-13:
> First is for not disconnecting USB drive during write = prevent UI_a112_badUSBDevice. Second for consistency of snapshot, there can be write in progress. Does this make sense?

Please submit your proposed changes through a pull request from your end.